supermileage/dynamometer-firmware #11: Interfacing with sd card reader on ili9341 display

cosparks, 2023-02-12:
- Add connection to ili9341 display with SPI or SPI1 for the pi (using chip select, you can have multiple devices on the same bus).
- Find library for reading and writing sd cards.
- Find appropriate library to use (for reading / writing sd cards)
- Find appropriate library for handling csv files (it might also be trivial to come up with your own code).

cosparks, 2023-02-12:
Related to #13

cosparks, 2023-02-12:
this work should branch from and eventually be merged into dev_pico

cosparks, 2023-03-05:
@underscoreHasan I'd also like to get an idea of performance: how long does it take to write a 1000 character string?  How long does it take to write a single column entry for a csv file?

Because we're going to have to potentially capture fairly high-resolution data for these csv tables we're building, I'd like to know whether or not we should be tasking the second core on the pico with some of this read/write stuff

underscoreHasan, 2023-03-07:
Noted, I'll implement time readings when I start working on the .csv writing logic again

cosparks, 2023-03-24:
@underscoreHasan bumping this issue.  If you're not sure how to go about doing this, then let me know.  Knowing how much processor time it takes to write to the sd card will inform how we should split work between the pico's two cores.

cosparks, 2023-03-28:
@LaPommeCosmique the comment I made above is still relevant too.  It would be nice to have some data on how long it takes to write small/big chunks of data to the SD card.

cosparks, 2023-03-29:
Closing this and opening new issue for sd performance testing

closed 2023-03-29 by cosparks