supermileage/dynamometer-firmware #13: Notes on testing

cosparks, 2023-02-12:
### Moment of Inertia Torque
- angular acceleration and velocity recorded at INTERVAL exported to micro sd as csv

### Brake Hysteresis Torque
- force and change in force (acceleration) as csv

### Losses
- Torque over angular velocity as csv
- need electric motor with known torque (or precision torque meter)

cosparks, 2023-02-12:
- Created because apparently people can't comment on milestones.  Please leave any comments here.
- please refer to https://github.com/supermileage/dynamometer-firmware/milestone/1 for related issues

LaPommeCosmique, 2023-02-12:
Thanks Colby for setting this up. Are you thinking of having three separate csv files for each type of test? Or having all the variables in one file?

cosparks, 2023-02-12:
@LaPommeCosmique my plan currently is to generate separate files for each test.  Each file would have a name with the type of test and number or something like that.  Makes it easier to distinguish between good and bad test data, rather than keeping all of it in a single file.

Example:
- MOI_Test1.csv , MOI_Test2.csv, etc..

LaPommeCosmique, 2023-02-13:
That sounds good, and I agree that will make testing more organized. How will we control file creation/naming, if everything is being done on the raspberry pi?  (I'm not too unfamiliar with this type of stuff, and I'm interested in how it'll work, but its not super important at the moment haha)

LaPommeCosmique, 2023-02-13:
Also just a comment on the csv files for losses - since losses depend on angular velocity, it would be useful to know the actual angular velocity during our tests. So I believe it would be better to have angular velocity, and torque (or force) as separate variables. Hope this makes sense!

cosparks, 2023-02-13:
> How will we control file creation/naming, if everything is being done on the raspberry pi?

Probably some generic name as a macro, then each time we start a test (by pushing a button), we add a new number to the end of the name.

Something like
```
#define TEST_NAME "MOI_test"

String currentTestName(TEST_NAME);
int currentTestNum = 0;

void loop() {
    if (startTest) {
        currentTestName.concat(currentTestNum++);
        
        ... do some stuff
    }
}

LaPommeCosmique, 2023-02-13:
okay awesome, that makes sense