supermileage/dynamometer-firmware #15: Logging class

cosparks, 2023-02-12:
- Build out class which encapsulates behaviour for buffering and writing live sensor data to disk
- Will need something to trigger start / stop recording of data
- depends on #11

cosparks, 2023-02-20:
@underscoreHasan I've assigned this issue to you as well, as it's pretty closely related to the one you're already working on.

Once you've tested the SD card reader, you can build out a class which we will use to buffer csv data and write it to disk.

cosparks, 2023-03-16:
I would also like the csv files we're writing to be rfc-compliant:
https://www.rfc-editor.org/rfc/rfc4180

cosparks, 2023-03-28:
@LaPommeCosmique hey Alex, since Hasan has left the team, you could pick this up and finish it.

Hasan has actually mostly completed it, but didn't follow the template I provided above 100%.  His work is on the branch dev_sdcard

LaPommeCosmique, 2023-04-01:
Here are the functions from the specs, mapped to the current functions that we have. 

```
bool create(String name, int numColumns)  -> bool createCSV(String path, vector<String> header)
bool open(String name)  -------------------> bool loadCSV(String path)
bool saveToDisk()  ------------------------> 
bool close()  -----------------------------> bool close()
void setHeader(String header)  ------------> setHeader(vector<String> header)
void addEntry(String data)  ---------------> 
void addRow(String data)  -----------------> void writeRow(vector<float> values)
int getNumColumns()  ----------------------> int getNumColumns()
int getBufferLength()  --------------------> 
-------------------------------------------> bool writeToLog(vector<float> values)
```

Currently
- all write functions take in vectors, and produce strings from those vectors (instead of taking in strings)
- all writes to the file are done immediately (instead of putting everything in a buffer)

Do you have any suggestions for what aspects you would like to keep and what aspects you want to change to match the original specifications?

LaPommeCosmique, 2023-04-01:
Worth mentioning that if we go with the original specifications, then we can use this class for testing (issue #29). Otherwise, I believe we have to create separate functions to test SD card read/writes.

cosparks, 2023-04-01:
I think everything should match the original specification.  Vectors add complexity and aren't necessary for what we're doing.  You could replace all those params with strings.

We could also pass a flag to the constructor which indicates whether or not we want every new row to be written directly to disk (in the C standard library this is called the O_SYNC flag), or buffered in memory and eventually written to disk (this could be the default setting).  This would help when doing those performance tests.

LaPommeCosmique, 2023-04-02:
Right I see. 

So this would be a string buffer right? And every time we add a line `line`, we would add `"\n" + line` to the buffer?

LaPommeCosmique, 2023-04-02:
I have added a couple of commits that should hopefully bring DataLogger up to spec. However I have not tested the code yet.

One (small) issue:
- Currently, the `open()` function cannot differentiate between a blank file, and a file with one header value. In both cases, it will calculate the number of columns as 0 (even though the latter should have 1 column). But I believe this is such an edge case that its almost falling off the edge.

LaPommeCosmique, 2023-04-02:
Also, just to confirm - are we using `String` instead of `string`?

cosparks, 2023-04-02:
> Also, just to confirm - are we using String instead of string?

Yep, I think it makes sense to keep using String as opposed to std::string (String is the arduino library's implementation)

> So this would be a string buffer right? And every time we add a line line, we would add line + "\n" to the buffer?

According to the [rfc for csv](https://www.ietf.org/rfc/rfc4180.txt) files, this should actually be 'carriage return line feed', which is "\r\n"

> Currently, the open() function cannot differentiate between a blank file, and a file with one header value. In both cases, it will calculate the number of columns as 0 (even though the latter should have 1 column). But I believe this is such an edge case that its almost falling off the edge.

I think it's good to consider cases like these.  We might want to have a way to clear a file, or just to detect that there is already a file with the same name and not the same column count, in which case we should make a new file with some suffix which differentiates the two

cosparks, 2023-04-02:
> edit: I have implemented a fix. However, currently when opening a csv file, it is possible that there are already many entries. Because of this, ideally we would want to add a newline if the previous character is not \n. However, I'm not sure if there is a way to read the last character of a file without going through the entire file first.

Checking this may not be necessary if we just assume that the last write to the file was formatted correctly (ends with CRLF).  I also don't think it would be too costly to read until the end of the file because timing isn't a huge priority during the setup state.  With or without the CRLF check, this flowchart kinda makes sense to me

[image]

LaPommeCosmique, 2023-04-03:
Okay sounds good, and thanks for the feedback. I have implemented the open() to follow the flowchart. A couple of comments
- `open(String name, int numColumns)` now takes in an additional `numColumns` parameter
- I am prefixing (instead of suffixing) `copyof_` to the file name if they already exist. This is we dont have to worry about interfering with file extensions 
- might be slightly pedantic but.. the algorithm will screw up if there is a header like:
``` header1, "header \n 2", header3 CRLF ```
since it is not taking into account quotation marks

Other than that I believe the DataLogger class should be up to spec 👍

cosparks, 2023-04-04:
> I am prefixing (instead of suffixing) copyof_ to the file name if they already exist. This is we dont have to worry about interfering with file extensions

I would actually parse the name and add a number as a suffix before the extension.  That way, we won't get super long file names like copyof_copyof_copyof_...file.csv if this is applied several times.  The number suffixing is also conventional for most applications that deal with files, like when you download two copies of a file with the same name in your browser.

> might be slightly pedantic but.. the algorithm will screw up if there is a header like: "header1, "header \n 2", header3 \r\n" CRLF--since it is not taking into account quotation marks

Interesting, but we should never wind up with quotation marks and random newlines in files like that.  If DataLogger produces files like that then it is broken, and if somebody tried to load a csv file formatted like that then it's kinda their problem

cosparks, 2023-04-04:
We also shouldn't have any spaces on the header line.  You can check this simple (and very short) [RFC document](https://www.ietf.org/rfc/rfc4180.txt) on csv files if you'd like.  If we expect our input to adhere strictly to this format then we don't have to worry to much about how to deal with edge cases.  Just display a failure notification (which you don't have to worry about now because we're not dealing with the UI yet)

cosparks, 2023-04-04:
Also, I think Hasan may have been using the auto-format command in vscode, which isn't standard with c++ style for elec.  I can change it, but if you'd like to know, here is elec's [c++ style guide](https://www.notion.so/C-Style-Guide-f40ef59a0c3140148e1e578bfc61f1b8)

closed 2023-06-16 by cosparks