supermileage/dynamometer-firmware #29: SD Card read/write performance

cosparks, 2023-03-29:
This is a pretty small issue which involves testing read and write performance of our spi sd card interface.  Goal is to answer the following questions:

- how long does it take to write a 1000 character string to the sd card?
- how long does it take to write a single column entry for a csv file?
- how long does it take to open a file and read 1000 bytes from sd card into a buffer?

Because we're going to have to potentially capture fairly high-resolution data for these csv tables we're building, I'd like to know whether or not we should be tasking the second core on the pico to perform these reads and writes

During a test session, we will also potentially be reading from one file while writing to (potentially multiple) others

LaPommeCosmique, 2023-03-31:
Since this will be independent from the actual firmware (if I understand correctly), do you suggest I create a new project or do everything on a separate branch?

cosparks, 2023-03-31:
Yes, this is just a one-off experiment.  You should branch from dev_sdcard (since Hasan already implemented much of the functionality you need to do this testing) and modify main.cpp to add some logic in the main loop that computes an average write time for n bytes to persistent storage.

I expect that write speed will be the minimum of the SPI baud and sd card write speed, but I'm not totally sure

LaPommeCosmique, 2023-04-02:
I have created a new branch called testing_sdcard and implemented some tests to look at this issue.

Only thing left (excluding debugging) is to add the PIN # of the SD card.

LaPommeCosmique, 2023-04-02:
Currently, all the results of the test are sent through DEBUG_SERIAL_LN(). Additionally, three files are created on the SD disk during testing - `test1.txt`, `test2.txt`, and `test3.txt`.

cosparks, 2023-04-02:
Cool, this looks great!

The pin can be any pin that isn't already being used on that pico (it's attached to a breadboard in the dyno box of the elec cabinet).

cosparks, 2023-04-02:
also, line 19 of DataLogger.cpp is problematic.

We would never want to loop infinitely if we fail to initialize a resource.  I think that Hasan added that, but you can remove it.  Printing the debug message is fine

LaPommeCosmique, 2023-04-03:
Good catch lol. I have removed the line

LaPommeCosmique, 2023-04-03:
Are you planning on running the tests with the pico, or were you wanting me to do them?

cosparks, 2023-04-04:
If you'd like to develop for this project I would definitely recommend grabbing the pico in the dyno box in the elec cabinet.  Seeing that your code compiles without error usually isn't enough to verify that it works

LaPommeCosmique, 2023-06-27:
Here are the results of the tests:

```
Test 1: how long does it take to write a 1000 character string to the sd card?
Average time: 15960.99 micros

Test 2: how long does it take to write a single column entry for a csv file? (one entry = "1.0")
Average time: 2578.43 micros

Test 3: how long does it take to write five column entries for a csv file? (osync = false)
Average time: 2469.09 micros

Test 4: how long does it take to open a file and read 1000 bytes from sd card into a buffer?
Average time: 19912.12 micros
```

In each test, I had 100 trials, and I took the average of all trials.

cosparks, 2023-07-03:
These are the final results after modifying the code to only create a file once at the start of each test:

```
Test 1: how long does it take to write a 1000 character string to the sd card?
Average time: 4286.59 micros

Test 2: how long does it take to write a single column entry for a csv file?
Average time: 32.82 micros

Test 3: how long does it take to write five column entries for a csv file (osync = false)?
Average time: 205.51 micros

Test 4: how long does it take to open a file and read 1000 bytes from sd card into a buffer?
Average time: 7777.85 micros
```

It turns out that the first write to disk after creating a new file takes quite a lot longer than subsequent writes.  The read and write speeds will also vary depending on the sd card.

PR #69 (2023-07-03): SD Performance Tests
- adds performance tests for reading and writing from sd card
- creates build environment for performance tests so they can be compiled and run separately from main application, which will also make it easier to add performance tests for other code if we wish
- additionally, from now on these tests will be built as part of this project's github workflows

closes #29
closed 2023-07-03 by cosparks