supermileage/dynamometer-firmware #42: Application State: Calibration display

cosparks, 2023-05-26:
It would be useful to have different input values and sensor readings displayed on the screen during calibration.  Things to consider are:
- timer
- current rpm / angular-acceleration / velocity and force (as a graph or just values)
- start, stop and other useful menu buttons
- path of csv file we are logging data to

Please post any sketches or design ideas for this UI in the comments below.

LaPommeCosmique, 2023-05-26:
Hey @cosparks we are exploring a ["gearless magnet" idea](https://docs.google.com/document/d/1W9JXqvRoIVGHvsfE__0bXLTGkSq_Wh_ixZVeQrBS5Vw/edit?usp=sharing) for calibration, where the motor is connected to a magnet wheel, and the shaft is connected to a copper disk. In this case torque is directly proportional to the speed differential of the magnet wheel and the copper disk.

If we go with this idea it would be useful to display the RPM of the motor (since it will be different from the RPM of the optical wheel) - I believe this we can get from VESC. And also, the difference in speed between the motor and shaft, as this would be proportional to torque.

cosparks, 2023-07-07:
This issue has ballooned quite a lot, as implementing this functionality well means building it out in such a way that it is reusable for the other 'session' states.

I have split it up into a number of issues, which I hope to finish in the following order:

#75 , #73, #74, #76, then finally this one.