supermileage/dynamometer-firmware #49: IO: Add BPM Control Class

cosparks, 2023-05-26:
There are multiple application states which will require the ability to control the BPM control voltage signal, so we need a class which encapsulates this behaviour.  Class should include:
- constructor which takes a pin number (type _pin_size_t_)
- _init_ or _begin_ method which initializes pin
- method: _setControlSignal(int32_t val)_ where val is a number from 0 to _N_
- method: _setActive(bool val)_ which 

Things to consider about _N_:
- _N_ is the analog write range on the pico, which needs to be set globally by calling _analogWriteRange_ in _setup_
- we also need to set the analog write frequency (aka pwm frequency) by calling _analogWriteFreq_ in _setup_
- there is some interplay between analog write range and frequency: [info](https://arduino-pico.readthedocs.io/en/latest/analog.html)
- the potentiometer analog read resolution is currently 10-bits (so has range [0-1023]), which means we could simply use the same range for the pwm.  We just need to make sure that we don't choose a pwm frequency so high that we can't achieve that resolution

Note that bpm ctrl output pin is GP3 (pin # 3)

cosparks, 2023-05-26:
This class can go in System folder

LaPommeCosmique, 2024-02-11:
Documentation for the physical BPM component: https://www.notion.so/Off-Board-Components-0a6a960854d7498bbabc718105479ab2

KevinZhiyuanDing, 2024-03-16:
I have made the BPM control class. 
The constructor takes the pin number for the BPM control output pin. 
I have added the pin number definition in ```settings.h```:
```
//BPM
#define BPM_CTRL_OUT        3
```
#include "HardwareInput.h"
#include "BpmControl.h"

BpmControl::BpmControl (pin_size_t pinNum){
    _pinNum = pinNum;
}

void BpmControl::init (){
    pinMode(_pinNum, OUTPUT);
    analogWriteRange();
    analogWriteFreq();
};

void BpmControl::setControlSignal (int32_t controlSignal){ 
    if (_active == true){
        if (controlSignal >= 0 && controlSignal <= 1023){
            analogWrite (_pinNum, controlSignal);
        }
    }
    else {
        analogWrite (_pinNum, 0);
    }
};

void BpmControl::setActive (bool active){
    _active = active;
    if (active == false) {
        analogWrite (_pinNum, 0);
    }
};
```
Its corresponding header file is:
```
#ifndef _BPM_CONTROL_H_
#define _BPM_CONTROL_H_
#include "settings.h"

#define ANALOG_MAX 1023;
#define ANALOG_MIN 0;

class BpmControl {
    private:    
        int _pinNum;
        bool _active = false;
    public:
        BpmControl (pin_size_t pinNum);
        void init ();
        void setControlSignal (int32_t signalStrength);
        void setActive (bool state);
};

#endif
```

closed 2024-09-04 by KevinZhiyuanDing