supermileage/dynamometer-firmware #62: Refactor: Remove all 64-bit numerical types from repo

cosparks, 2023-06-25:
The RP2040 is a 32-bit microcontroller.  It can handle operations on 64-bit integers and floating point values, but it is time-consuming and probably unnecessary given the scope of this project.

The goal with this issue is to simply remove all 64-bit types and replace them with the appropriate signed/unsigned/floating point types.

Here is a chart of different durations of double operations vs float operations (floating point operations have _f_ in the name whereas double operations have _d_; we are using the ROM/SDK floating point functions):
[image]

If there are any concerns, please post them below.

LaPommeCosmique, 2023-06-25:
Seems like a pretty easy change to achieve the increased performance benefits. I can take care of this. I will globally replace:
- ```double``` to ```float```
- ```u_int``` to ```u_int32_t```
- ```int``` to ```int32_t```

Would that be good?

LaPommeCosmique, 2023-06-26:
Oh, according to this link - https://raspberry-projects.com/pi/programming-in-c/memory/variables - it looks like its only doubles that we have to worry about 
[image]

cosparks, 2023-06-26:
> Seems like a pretty easy change to achieve the increased performance benefits. I can take care of this. I will globally replace:
>     * `double` to `float`
>     * `u_int` to `u_int32_t`
>     * `int` to `int32_t`

uint is a pico sdk defined type, which maps to unsigned int, which is a 32-bit integer.  The size of int / unsigned int and all the other normal C int types vary depending on the architecture you're using, so it's best to use the explicit int types from <stdint.h>, where the size will be same across all architectures.

I think the only places that I use int / uint are places where an sdk function is consuming or returning a value of that type.

closed 2023-06-26 by cosparks