supermileage/dynamometer-firmware #63: Refactor: Consider usage of micros()

cosparks, 2023-06-25:
The micros() function returns a 32-bit unsigned integer value representing the time in microseconds since the microcontroller booted up.  We are using it in at least one place (in src/Sensor/SensorOptical.cpp).  The potential hazard here is that the largest value which can be represented with an unsigned 32-bit int is 2^32-1, so every ~71.5 minutes, micros will wrap (ie. will start counting from zero).

The question is whether or not we actually need microsecond precision for averaging the frequency of the signal we are getting from the optical sensor, which will be at the very maximum ~25kHz.  In any case, the averaging code in SensorOptical will need to be modified to either:

- replace micros with millis, which will wrap every 50 days instead (in which case we won'just t need to handle wrapping)
- add logic to handle micros wrapping around to 0 every 2^32-1 microseconds

cosparks, 2023-06-25:
@LaPommeCosmique just bringing this to your attention

LaPommeCosmique, 2023-06-25:
I think its tough to compare the performance of having ~25kHz resolution compared to having ~1kHz resolution (using the ```millis()``` function) without some testing. Maybe the performance differences between the two will be negligible, but its hard to say right now. Because of this, I would support the second option (to add logic to handle micros wrapping) so we can avoid sacrificing resolution.

cosparks, 2023-06-26:
I'm not entirely sure what you mean.  25kHz is the max frequency of the optical sensor signal if we are running a vehicle at 80kmph or something ridiculous.  Switching from micros to millis just means that the shortest interval on which we can compute and log angular velocity / RPM is 1ms as opposed to ~10us (or greater because I imagine our control loop is already longer than a few microseconds).

It's more an issue of the resolution of the data we are collecting and storing in a csv.  It won't have a big effect on performance I don't think.

cosparks, 2023-06-26:
Using micros will yield higher-precision values when it comes to actually computing the average angular velocity over short intervals.  I'm not sure how short these intervals are going to be or how much precision we really gain from that, but it may be worthwhile still.

I support your decision if you want to keep using micros.  The extra instructions for handling wrapping shouldn't be too cumbersome I don't think.

LaPommeCosmique, 2023-06-26:
> It's more an issue of the resolution of the data we are collecting and storing in a csv. It won't have a big effect on performance I don't think.

Ah my bad. When I was talking about "performance", I was referring to the performance in terms of how precise we could measure velocity/acceleration. "Precision" would have been a better word than "performance".

> I'm not sure how short these intervals are going to be or how much precision we really gain from that, but it may be worthwhile still.

I agree - this is my reasoning too. Since we currently can't quantify the precision loss by using ```millis()``` compared to ```micros()```, its better to not sacrifice resolution right now in case there is actually a noticeable difference in precision.

closed 2023-07-02 by cosparks