supermileage/dynamometer-firmware #83: HardwareRotaryEncoder possibly allows for skipping of values

LaPommeCosmique, 2023-09-04:
Here is a code snippet from [HardwareRotaryEncoder.h](https://github.com/supermileage/dynamometer-firmware/blob/dev/src/System/HardwareRotaryEncoder.h)

```
 void run() override {
    PinStatus valA = digitalRead(_pinA);
    
    if (valA != _lastA) {
        PinStatus valB = digitalRead(_pinB);    
        
        if (valA && valB) {
            ++_counter;
        } else if (valA && !valB) {
            --_counter;
        }
        _lastA = valA;
    }

    /* more code */
}
```

It appears that if someone wiggles the rotary encoder so that pin A oscillates between `ON` and `OFF`, and pin B remains constant (either `ON` or `OFF`), then the rotary encoder will experience a phantom rotation even though it is merely wiggling back and forth.

| `valA` | `valB` | `_lastA` | `counter` | |
|--------|--------|--------|--------|--------|
| FALSE | FALSE | FALSE | 0 | initial state |
| TRUE | FALSE | FALSE | -1 | encoder is rotated a bit so that pin A is triggered |
| FALSE | FALSE | TRUE| -1 | encoder is rotated back to its original state |
| TRUE | FALSE | FALSE | -2 | encoder is rotated again so that pin A is triggered  |

cosparks, 2023-11-07:
@LaPommeCosmique I can't remember whether we talked about this.  I tested this behaviour and it's really not bad (I think I've had some synthesizers which have done the same thing).  You have to very intentionally move the rotary knob to reproduce.

It might be a good introductory issue for this codebase, but not super high priority

KevinZhiyuanDing, 2023-11-11:
So James and I found a possible small issue with the original rotary encoder logic. If the rotary encoder follows this convention: 
[image]
where inputA is 90 degrees out of phase with inputB and input A goes first, then the original code:

if (valA != _lastA) {
                PinStatus valB = digitalRead(_pinB);    
           
                if (valA && valB) {
                    ++_counter; /ccw
                
                } else if (valA && !valB) {
                    --_counter; /cw
                }

only considers situations where A is originally on a logic low (0) and goes to logic high (1). 
If A is originally on a logic high (1) and goes to a logic low (0), it isn't considered. 
Therefore, wiggling the rotary encoder in place such that A oscillates in place between (1) and (0) simulates a rotation even though there isn't one.

The fix we came up with is this:

void run() override {
            PinStatus valA = digitalRead(_pinA);
            
            if (valA != _lastA) {
                PinStatus valB = digitalRead(_pinB);    
                
                if (valA != valB){
                    ++_counter; //done when rotating counterclockwise
                }
                else if (valA == valB){
                    --_counter; //done when rotating clockwise
                }

                /* Original
                if (valA && valB) {
                    ++_counter; /ccw
                
                } else if (valA && !valB) {
                    --_counter; /cw
                }
                */
                _lastA = valA;
            }

closed 2024-02-17 by LaPommeCosmique